nxp-imx/mfgtools #436: uuu 1.5.182 doesn't build on Fedora 39

tprrt, 2024-10-02:
Following issue occurs on Fedora 39 when I'm trying to build uuu 1.5.182:
```
cmake . && make
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Warning (dev) at libuuu/CMakeLists.txt:65:
  Syntax Warning in cmake code at column 159

  Argument not separated from preceding token by whitespace.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Deprecation Warning at libuuu/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 9.0.0
-- Found OpenSSL: /usr/lib64/libcrypto.so (found version "3.1.4")  
CMake Deprecation Warning at uuu/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Checking for module 'zlib'
--   Found zlib, version 1.2.13
-- Configuring done (0.6s)
-- Generating done (0.0s)
-- Build files have been written to: /src
[  2%] Generating gitversion.h
build not in appveyor
mv: cannot stat '/src/libuuu/gen/gitversion.h.tmp': No such file or directory
make[2]: *** [libuuu/CMakeFiles/uuc_s.dir/build.make:75: libuuu/gitversion.h] Error 1
make[1]: *** [CMakeFiles/Makefile2:115: libuuu/CMakeFiles/uuc_s.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

```

nxpfrankli, 2024-10-02:
we prefer build from git repo. Did you install git?

tprrt, 2024-10-02:
Hello,

> we prefer build from git repo. Did you install git?

With Git sources, the build is successful.
There seem to be some missing files in the tag’s source tarball.

Kind regards,
Thomas