nxp-imx/mfgtools #438: Memory corruption when using python binding (Mac OS X AppleSilicon)

mzachar, 2024-10-17:
I'm using python wrapper for libuuu  to execute `FB: getvar version` but the response is corrupted. It looks like either the buffer is overridden or there is some issue with handling null byte terminated string.

Sample python program:

```python
import libuuu

if __name__ == '__main__':
    print("libuuu version: ", libuuu.get_libuuu_version())

    uuu = libuuu.LibUUU()
    print("uuu version: ", uuu.get_version_string())

    for i in range(1,6):
        uuu.run_cmd("FB: getvar version", False)
        print(f"{i} FB: getvar version: ", uuu.response)
```

will output:

```
libuuu version:  1.5.182
uuu version:  uuu_1.5.182-0-g18656db
1 FB: getvar version:  b'0.4\x1f{\xb7)\x88\x08'
2 FB: getvar version:  b'0.4\x1f{\xb7)x\x08'
3 FB: getvar version:  b'0.4versionE\x88\x02h'
4 FB: getvar version:  b'0.4\x1f{\xb7)\x88\x08'
5 FB: getvar version:  b'0.4'

Process finished with exit code 0
```

nxpfrankli, 2024-10-17:
@UUUNotifyCallback
def _default_notify_callback(struct: UUUNotifyStruct, data) -> int:  # type: ignore
    """A default callback function that stores the response in a class variable.

    :param struct: A UUUNotifyStruct object
    :param data: A pointer to data, here it is not used
    """
    # pylint: disable=unused-argument
    LibUUU._state.update(struct)
    if struct.type == UUUNotifyType.NOTIFY_CMD_INFO:
        LibUUU._response.value += bytes(struct.response.str)

Suppose problem should be here ^^^

    return 1 if LibUUU._state.error else 0

mzachar, 2024-11-02:
I just tested this on Windows 11 and it works as expected:

```
libuuu version:  1.5.182
uuu version:  uuu_1.5.182-0-g18656db
1 FB: getvar version:  b'0.4'
2 FB: getvar version:  b'0.4'
3 FB: getvar version:  b'0.4'
4 FB: getvar version:  b'0.4'
5 FB: getvar version:  b'0.4'
```