nxp-imx/mfgtools #440: "Synchronous Abort" handler, esr 0x96000000

lynchwy, 2024-10-18:
When debugging the UUU function of the IMX8mm chip, the following error occurs:
[spl_sdp_load_image:29] g_dnl_clear_detach

[g_dnl_register:296] g_dnl_driver.name = usb_dnl_sdp

[usb_composite_register:1499] usb_gadget_register_driver usb_dnl_sdp

[usb_gadget_register_driver:1008]

[usb_gadget_register_driver:1022] ci_udc_probe

"Synchronous Abort" handler, esr 0x96000000

elr: 00000000007e5714 lr : 00000000007e627c

x0 : 0000000042200010 x1 : 00564000defd7342

x2 : 00000000007fc000 x3 : 0000000042200000

x4 : 0000000042200000 x5 : 00564001211d7342

x6 : 0000000000000000 x7 : 0000000042201340

x8 : 00000000007fc428 x9 : 0000000000000002

x10: 00000000ffffffd8 x11: 000000000000000d

x12: 0000000000000006 x13: 0000000000000008

x14: 000000000090e558 x15: 0000000000013da4

x16: 0000000000000000 x17: 0000000000000005

x18: 000000000091de40 x19: 0000000042200ff0

x20: 0000000000001000 x21: 0000000000000310

x22: fffffffffffff010 x23: 0000000000000300

x24: 00000000007f24c0 x25: 00000000007f3451

x26: 00000000007f3431 x27: 00000000007f22d8

x28: 0000000000000000 x29: 000000000091dc80

Resetting CPU ...

imx-p9.0.0_2.3.4.tar.gz uboot-imx source code
imx8mm_evk_android_defconfig

nxpfrankli, 2024-10-18:
it should uboot issue.  Maybe image try to download overwrite uboot execute code section

lynchwy, 2024-10-18:
Any suggestions for me to troubleshoot this issue?