nxp-imx/mfgtools #446: Null pointer when flashing a sdimg.bz2 image

jakakordez, 2024-12-05:
When I try to flash my board using a .sdimg.bz2 image, the flashing almost every time halts. I cloned the project and ran it with Visual Studio and I found out that there is a null pointer in the g_filebuffer_map of libuuu/buffer.cpp which causes a read access violation. I verified that null is never stored into the map explicitly in buffer.cpp:1211.

[image]
[image]

My system is Windows 11 and I am flashing an i.MX8M Plus board.
The problem is also that when uuu.exe is ran directly, without a debugger attached and that happens, the flashing procedure just freezes without any error message to the user.

nxpfrankli, 2024-12-05:
What's size of your bz2? And how do you create bz2, by pbzip2 or bzip2?  Suggest switch to use zstd.

jakakordez, 2024-12-06:
The original sdimg is 7GB, compressed to bz2 is 137 MB. Image is produced using the Yocto build so I am not sure which compression tool is used internally.