nxp-imx/mfgtools #455: Unknown exit code -1073740791

Sergi-CP, 2025-02-13:
Hi,
I need your help to find out what can be hapenning flashing my device (MX8QXP). Always works perfect but I've a "drip" of flashing chips with this exit code. I try to find this error code but not success. If I take a look to the process all it's done ok:
 
0 (Uuu.exe has been terminated with ExitCode = -1073740791 - ConsoleOutput = uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.182-0-gda3cd53
Your console don't support VT mode, fail back to verbose mode
Build in config:
Pctl Chip Vid Pid BcdVersion Serial_No
==================================================
SDPS: MX8QXP 0x1fc9 0x012f [0x0002..0xffff]
SDPS: MX8QM 0x1fc9 0x0129 [0x0002..0xffff]
SDPS: MX8DXL 0x1fc9 0x0147
SDPS: MX28 0x15a2 0x004f
SDPS: MX815 0x1fc9 0x013e
SDPS: MX865 0x1fc9 0x0146
SDPS: MX8ULP 0x1fc9 0x014a
SDPS: MX8ULP 0x1fc9 0x014b
SDPS: MX93 0x1fc9 0x014e
SDPS: MX91 0x1fc9 0x0159
SDPS: MX95 0x1fc9 0x015d
SDPS: MX95 0x1fc9 0x015c
SDP: MX7D 0x15a2 0x0076
SDP: MX6Q 0x15a2 0x0054
SDP: MX6D 0x15a2 0x0061
SDP: MX6SL 0x15a2 0x0063
SDP: MX6SX 0x15a2 0x0071
SDP: MX6UL 0x15a2 0x007d
SDP: MX6ULL 0x15a2 0x0080
SDP: MX6SLL 0x1fc9 0x0128
SDP: MX7ULP 0x1fc9 0x0126
SDP: MXRT106X 0x1fc9 0x0135
SDP: MX8MM 0x1fc9 0x0134
SDP: MX8MQ 0x1fc9 0x012b
SDPU: SPL 0x0525 0xb4a4 [0x0000..0x04ff]
SDPV: SPL1 0x0525 0xb4a4 [0x0500..0x9998]
SDPV: SPL1 0x1fc9 0x0151 [0x0500..0x9998]
SDPU: SPL 0x0525 0xb4a4 [0x9999..0x9999]
SDPU: SPL 0x3016 0x1001 [0x0000..0x04ff]
SDPV: SPL1 0x3016 0x1001 [0x0500..0x9998]
FBK: 0x066f 0x9afe
FBK: 0x066f 0x9bff
FBK: 0x1fc9 0x0153
FB: 0x0525 0xa4a5
FB: 0x18d1 0x0d02
FB: 0x3016 0x0001
FB: 0x1fc9 0x0152
FB: 0x0483 0x0afb
Wait for Known USB Device Appear... at path 2:4
New USB Device Attached at 2:4-1616100E8294597B
2:4-1616100E8294597B>Start Cmd:SDPS: boot -f imx-loader
3%
6%
9%
12%
15%
18%
22%
25%
28%
31%
34%
37%
41%
44%
47%
50%
53%
56%
60%
63%
66%
69%
72%
75%
79%
82%
85%
88%
91%
94%
98%
100%2:4-1616100E8294597B>Okay (2.792s)
New USB Device Attached at 2:4-1616100E8294597B
2:4-1616100E8294597B>Start Cmd:FB: ucmd setenv fastboot_dev mmc
2:4-1616100E8294597B>Okay (0.151s)
2:4-1616100E8294597B>Start Cmd:FB: ucmd setenv fastboot_buffer ${loadaddr}
2:4-1616100E8294597B>Okay (0.004s)
2:4-1616100E8294597B>Start Cmd:FB: ucmd setenv mmcdev ${emmc_dev}
2:4-1616100E8294597B>Okay (0.004s)
2:4-1616100E8294597B>Start Cmd:FB: ucmd mmc dev ${emmc_dev}
2:4-1616100E8294597B>Okay (0.153s)
2:4-1616100E8294597B>Start Cmd:FB: download -f imx-boot
2:4-1616100E8294597B>Okay (0.629s)
2:4-1616100E8294597B>Start Cmd:FB: ucmd mmc dev 0 1
2:4-1616100E8294597B>Okay (0.155s)
2:4-1616100E8294597B>Start Cmd:FB: ucmd mmc write ${loadaddr} 0 0x5d6b
2:4-1616100E8294597B>Okay (0.203s)
2:4-1616100E8294597B>Start Cmd:FB: ucmd mmc partconf ${emmc_dev} 0 1 0
2:4-1616100E8294597B>Okay (0.007s)
2:4-1616100E8294597B>Start Cmd:FB: flash gpt file1.bin
0x400000002:4-1616100E8294597B>Okay (0.331s)
2:4-1616100E8294597B>Start Cmd:FB: ucmd mmc dev 0
2:4-1616100E8294597B>Okay (0.153s)
2:4-1616100E8294597B>Start Cmd:FB: ucmd env set fastboot_buffer 0x99000000
2:4-1616100E8294597B>Okay (0.004s)
2:4-1616100E8294597B>Start Cmd:FB: download -f file1.ext4.lz4
2:4-1616100E8294597B>Okay (6.923s)
2:4-1616100E8294597B>Start Cmd:FB[-t 30000]: ucmd ulz4tommc 0x800
2:4-1616100E8294597B>Okay (16.42s)
2:4-1616100E8294597B>Start Cmd:FB: download -f file1-varsysext4.lz4
2:4-1616100E8294597B>Okay (0.019s)
2:4-1616100E8294597B>Start Cmd:FB[-t 3000]: ucmd ulz4tommc 0x258800
2:4-1616100E8294597B>Okay (1.32s)
2:4-1616100E8294597B>Start Cmd:FB: download -f file1-hmi.ext4.lz4
2:4-1616100E8294597B>Okay (15.51s)
2:4-1616100E8294597B>Start Cmd:FB[-t 30000]: ucmd ulz4tommc 0x271800
2:4-1616100E8294597B>Okay (21.16s)
2:4-1616100E8294597B>Start Cmd:FB: download -f version.raw
2:4-1616100E8294597B>Okay (0.007s)
2:4-1616100E8294597B>Start Cmd:FB: ucmd mmc write ${fastboot_buffer} 0x5f5800 4
2:4-1616100E8294597B>Okay (0.008s)
2:4-1616100E8294597B>Start Cmd:FB: download -f file1-userdata.ext4.lz4
2:4-1616100E8294597B>Okay (0.065s)
2:4-1616100E8294597B>Start Cmd:FB[-t 15000]: ucmd ulz4tommc 0x5f6000
2:4-1616100E8294597B>Okay (6.49s)
2:4-1616100E8294597B>Start Cmd:FB: done
2:4-1616100E8294597B>Okay (0s)
For more information, refer to the log file.)

Would you be so kind to check why I can have this exit code please. MANY THANKS FOR ADVANCE!

nxpfrankli, 2025-02-14:
I look like crash when do cleanup work when finish task. Does crash  every time?

Sergi-CP, 2025-02-14:
Thanks a lot for your reply, no it's not every time, it's alternating it flashes 2-3 times ok 1-2 with this exit code, I compared the output code between a good part and a bad part and only I see the difference that the good part ends the report with [53?

Sergi-CP, 2025-02-17:
Hi again, If it's a cleanup work can I supose that this board is OK?