nxp-imx/mfgtools #463: Does uuu Support flash real time image to eMMC

bkb2018, 2025-03-13:
I hava download a Pre-Built image from [Real-Time Edge Software | NXP Semiconductors](https://www.nxp.com/design/design-center/software/development-software/real-time-edge-software:REALTIME-EDGE-SOFTWARE) for iMX9, but use uuu command below can not flash the imx-boot-sd.bin-flash_singleboot into eMMC

uuu.exe -b emmc imx-boot-sd.bin-flash_singleboot

it report errors in windows: LIBUSB_ERROR_PIPE(-9).

What should i do to resove this problem.

Regards

nxpfrankli, 2025-03-13:
May it include v2x image, can you try last uuu after 1.5.201

bkb2018, 2025-03-14:
> May it include v2x image, can you try last uuu after 1.5.201

Did you mean i should git clone the latest mfgtools.git and build the uuu.exe tools by myself with vs2017 and then try again the flash flow.

bkb2018, 2025-03-14:
> May it include v2x image, can you try last uuu after 1.5.201

I had git clone the latest mfgtools.git and build uuu.exe with vs2017, but it can not work as well.

nxpfrankli, 2025-03-14:
Any log from uart console?

bkb2018, 2025-03-17:
> Any log from uart console?

Run built-in script:

uuu_version 1.2.39

# @_flash.bin            | bootloader
# @_image   [_flash.bin] | image burn to emmc, default is the same as bootloader

# This command will be run when i.MX6/7 i.MX8MM, i.MX8MQ
SDP: boot -f imx-boot-imx91-11x11-lpddr4-evk-sd.bin-flash_singleboot

# This command will be run when ROM support stream mode
# i.MX8QXP, i.MX8QM
SDPS: boot -f imx-boot-imx91-11x11-lpddr4-evk-sd.bin-flash_singleboot

# These commands will be run when use SPL and will be skipped if no spl
# SDPU will be deprecated. please use SDPV instead of SDPU
# {
SDPU: delay 1000
SDPU: write -f imx-boot-imx91-11x11-lpddr4-evk-sd.bin-flash_singleboot -offset 0x57c00
SDPU: jump
# }

# These commands will be run when use SPL and will be skipped if no spl
# if (SPL support SDPV)
# {
SDPV: delay 1000
SDPV: write -f imx-boot-imx91-11x11-lpddr4-evk-sd.bin-flash_singleboot -skipspl
SDPV: jump
# }

FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev ${emmc_dev}
FB: ucmd mmc dev ${emmc_dev}
FB: flash bootloader imx-boot-imx91-11x11-lpddr4-evk-sd.bin-flash_singleboot
FB: ucmd if env exists emmc_ack; then ; else setenv emmc_ack 0; fi;
FB: ucmd mmc partconf ${emmc_dev} ${emmc_ack} 1 0
FB: Done

Wait for Known USB Device Appear...
New USB Device Attached at 1:11-B1C5114598464CD4
1:11-B1C5114598464CD4>Start Cmd:SDPS: boot -f imx-boot-imx91-11x11-lpddr4-evk-sd.bin-flash_singleboot
4%1:11-B1C5114598464CD4>Fail HID(W): LIBUSB_ERROR_PIPE (-9)(0.033s)

Did log like this help?

bkb2018, 2025-03-22:
Sorry, the SoC is iMX91P not iMX91, Did uuu and the Pre-Built image support this?

bkb2018, 2025-03-22:
> Any log from uart console?

Sorry, the SoC is iMX91P not iMX91, Did uuu and the Pre-Built image support this?

bkb2018, 2025-03-27:
> Any log from uart console?

I have connected two usb cable to board, one for serial download and one for debug uart port, while use uuu to flash the image, the debug uart port had no any logs.

nxpfrankli, 2025-03-27:
Look like prebuild image don't match your board, ROM direct reset chips