nxp-imx/mfgtools #469: LIBUSB Error after running SDP command.

benlodz, 2025-04-03:
I'm working with a MX7D chip and after running `SDP: boot -f u-boot.imx`, it will go to 100%. However, attempting to run any other command following that will net me a `HID (W): LIBUSB_ERROR_IO (-1)` error. 

I built uuu from the latest commit on Fedora 41. I can program the chip on Windows and I don't really get any more details beyond that when running with both `-v` and `-V`. I believe I encountered the same issue when running the program on Ubuntu from the apt repo. 

I'm more than happy to provide any relevant logs or screenshots to fix this. Thanks.

allenxh, 2025-04-03:
Any log from board UART console?

Sincerely,

Han XU

benlodz, 2025-04-07:
I don't have any logs from the UART console. It's in recovery / serial mode so I'm not really sure how to get any logs from the device. Is there any relevant commands or a way to interact the device with picocom? 

Best,
Ben

benlodz, 2025-04-08:
Closing this issue as the solution was to pass the -nojump options.

closed 2025-04-08 by benlodz