nxp-imx/mfgtools #471: Devices appears as SCSI after SPD jump

benlodz, 2025-04-08:
I'm closing in on finally getting my uuu script to work. I'm currently migrating from a ucl2.xml and currently I got this so far

```sh
SPD: dcd -f u-boot.imx
SPD: write -f "u-boot.imx" -ivt 0
SPD: write -f "zImage" -addr 0x80800000
SPD: write -f "initramfs" -addr 0x83800000
SPD: write -f "device_tree_file" -addr 0x83000000
SPD: jump -f "u-boot.imx" -ivt 0
```

All of these commands do work, however the board comes up as SCSI device for USB block storage. Looking at dmesg, I get complaints that it's not having IO errors.  My understanding is now, I should be able to use FBK commands. Any guidance on this? Thank you.

FYI, the initramfs and device tree file are placeholders and are much longer. 

Best,
Ben

nxpfrankli, 2025-04-08:
You need update your initramfs with latest https://github.com/nxp-imx/imx-uuc

benlodz, 2025-04-09:
I'm not sure if it's the case with the initramfs. I examined both the ucl2.xml file, I'm migrating from and the device itself after it connects. 

The vid is: 0x066F and the pid is 0x37FF. There is a match for the vid but not the pid. What would be the workaround here or is uuu unable to support such a device?

nxpfrankli, 2025-04-09:
UUU use different protocol with old mfgtool.  So have to update initramfs

closed 2025-04-22 by benlodz