nxp-imx/mfgtools #472: U-Boot error because of wrong USB endpoint

dolcini, 2025-04-14:
With U-Boot 2025.04, i.MX8M Plus SoC, and latest UUU release, there are errors during download using fastboot protocol.

```
Starting download of 1842 bytes
dwc3-generic-peripheral usb@38100000: request 00000000ffb29080 was not queued to ep1in-bulk
downloading of 1842 bytes finished
dwc3-generic-peripheral usb@38100000: request 00000000ffb29080 was not queued to ep1in-bulk
dwc3-generic-peripheral usb@38100000: request 00000000ffb29080 was not queued to ep1in-bulk
Starting download of 118 bytes
dwc3-generic-peripheral usb@38100000: request 00000000ffb29080 was not queued to ep1in-bulk
downloading of 118 bytes finished
dwc3-generic-peripheral usb@38100000: request 00000000ffb29080 was not queued to ep1in-bulk
dwc3-generic-peripheral usb@38100000: request 00000000ffb29080 was not queued to ep1in-bulk
Starting download of 43289099 bytes
dwc3-generic-peripheral usb@38100000: request 00000000ffb29080 was not queued to ep1in-bulk
..........................................................................
..........................................................................
..........................................................................
..........................................................................
..................................
```

The download succeed despite these errors.

The issue seems related to some hard-coded USB endpoints in the UUU tool.

More details available here https://lore.kernel.org/all/20250319220805.219001-1-marex@denx.de/ and https://lore.kernel.org/all/Z_TnLsKHod_fvDsK@gaggiata.pivistrello.it/

Cc: @marex @nxpfrankli @MrVan 

[verdin-imx8mp-uuu-issue.txt](https://github.com/user-attachments/files/19732462/verdin-imx8mp-uuu-issue.txt)

nxpfrankli, 2025-04-14:
can you try mfgtools branch ep_in?  dymantic search  interrupt endpoint number.

dolcini, 2025-04-15:
@nxpfrankli 

tested `uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.201-11-g6df71aaa0789`

with latest U-Boot master, same error

```
# uuu recovery
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.201-11-g6df71aaa0789

Success 1    Failure 0                                                
                                                                       
                                                                       
3:111-154128 2/ 2 [Done                                  ] FB: done                                                     
```

```
Starting download of 16384 bytes
dwc3-generic-peripheral usb@31000000: request 0000000099f27040 was not queued to ep1in-bulk

downloading of 16384 bytes finished
dwc3-generic-peripheral usb@31000000: request 0000000099f27040 was not queued to ep1in-bulk
```

nxpfrankli, 2025-04-15:
It should be other issue